IUNO-TDM/PumpControl #29: add support for GPIOs

bockha, 2018-01-12:
The new [MotorShield](https://github.com/IUNO-TDM/MotorShield/tree/V01.00.00) contains three input connectors and one output connector. The inputs are thought to be connected to a _startbutton_ and door switches, the output is thought to control the illumination of the _startbutton_.

The pump control shall be enhanced to publish changes of the inputs using the websocket interface. It shall also be enhanced to enable the control of the startbutton illumination over the websocket.

bockha, 2018-02-02:
[article about internal pull_up and pull_down](http://raspi.tv/2013/rpi-gpio-basics-6-using-inputs-and-outputs-together-with-rpi-gpio-pull-ups-and-pull-downs)

[internal pullup using pigpio](http://abyz.me.uk/rpi/pigpio/python.html#set_pull_up_down)

goergch, 2018-02-08:
for MixerControl #117 we need a interface in the pumpcontrol to read and write GPIO Pins.

reading and writing should be possible via the REST API. There should also be events, triggered by changing read signals. this should be implemented in the websocket interface

Configuration and API design should be discussed before implementing the feature

goergch, 2018-02-27:
The user of PumpControl (like the MixerControl) should not know the GPIO PIN numbers. These should be configured in a settings file. E.g.:
gpio.0.pin = 4
gpio.0.name = cabinet_door_switch
gpio.0.direction = input
gpio.0.pullup = true

gpio.1.pin = 22
gpio.1.name = juice_door_switch
gpio.1.direction = input
gpio.1.pullup = true

gpio.2.pin = 23
gpio.2.name = start_button
gpio.2.direction = input
gpio.2.pullup = true

gpio.3.pin = 24
gpio.3.name = start_button_illumination
gpio.3.direction = output

goergch, 2018-02-27:
The REST interface should be extended by the following routes:

for output-pins:
PUT /gpio/(gpio-name)/outputenabled  body:(true / false)
GET /gpio/(gpio-name)/outputenabled returns:(true / false)

for input-pins:

GET /gpio/(gpio-name)/inputvalue returns:(true / false)

goergch, 2018-02-27:
The websocket interface should send messages, when an input value changes or on initial connection.
Message format:
gpio: {
  pin: 4
  name: start_button
  value: false
}

commit af0bd4d8 (2018-03-06): Closes #29: final cosmetics
closed 2018-03-06 by meierha